MrYsLab/pymata4 #5: Feature Request: Firmata over network

d-a-v, 2020-05-07:
Hello!

I can't find any reference in constructors on how to work on a network link with for example [StandardFirmataWiFi](https://github.com/firmata/arduino/tree/master/examples/StandardFirmataWiFi) (or FirmataEthernet in the same directories).

Is it a planned feature ?

MrYsLab, 2020-05-07:
Thanks for the question. At this point in time, pymata4 only works with StandardFirmata or FirmataExpress. 
I have no immediate plans in supporting StandardFirmataWifi for Arduino boards, not because I do not wish to, but because I have no way of testing it if I do. 
Which board(s) specifically are you looking for support? If it one that I own, I would certainly consider a WiFi implementation.

d-a-v, 2020-05-07:
I flashed the StandardFirmataWiFi configured as server, on a generic esp8266.
It would be very interesting to have it with python !
To check it works I tried successfully with a js client.

MrYsLab, 2020-05-07:
I will consider doing it for the esp8266 since this is something I can test here. I cannot promise a release date, but I should be able to take a look to see what the scope of the work is in the next week or so. I will post here when I have something to report. If  you don't hear from me by May 15, please ping by adding a comment here.

MrYsLab, 2020-05-08:
@d-a-v If your goal is to be able to control the ESP-8266 over WiFi using Python, I already have a solution available. It entails using [python_banyan](https://mryslab.github.io/python_banyan/#) and its [OneGPIO messaging protocol specification.](https://mryslab.github.io/python_banyan/#gpio_intro/).

There is a fully working demo available using a tkinter GUI to control the ESP-8266.

[image]
On the ESP side, it uses a [custom MicroPython script](https://github.com/MrYsLab/python_banyan/blob/master/projects/OneGPIO/esp_8266/esp_8266_micropython/esp_8266Full.py) that implements a TCP socket server.

Replacing the tkinter code with a straight Python script is very straight forward and I would be happy to explain things if you wish.

MrYsLab, 2020-05-08:
Could you please let me know which J  client you used and the JS script you used to test things.
Thanks.

d-a-v, 2020-05-08:
@MrYsLab Thanks for the tip !
I came here following the Firmata path, trying to stick to the most widely used protocols or standards (btw the link to Pymata from there should be updated).
As being an esp8266-Arduino maintainter, I mainly focus on arduino solutions but this is good to know !

For JS using node-js, I followed this instructable https://www.instructables.com/id/ESP8266-Firmata-J5-NodeBot/
For JS using a web browser, Websocket protocol is needed on Firmata side. I already asked Firmata folks about what they think of such an example.

MrYsLab, 2020-05-08:
Thanks. I will try things out first with the JS stuff and then see how to adapt pymata4. 
BTW, another shameless plug, the advantage of using the OneGPIO spec is that one can create a single Python script and use the same script to control an ESP-8266, Raspberry Pi, or Arduino without any code changes (except perhaps for pin numbers). Because it runs as a distributed process, you can even run multiple types simultaneously from the same script.

If I am successful in adapting pymata4 for WiFi, I will then consider creating a OneGPIO gateway for the ESP that uses Firmata similarly in the same fashion as the Arduino Gateway. Maybe the best of both worlds ;-)

MrYsLab, 2020-05-09:
An update. I will start working on this sometime next week. I need to prototype the socket interface. I will update when progress has been made (either postiive or negative).

d-a-v, 2020-05-09:
Thank you for these updates !
I made a very simple setup with a servo + a rotative servo in a lego-compatible box, with an esp01 + regulator + 18650. Kids will be able to pilot a carobot directly with your library !

MrYsLab, 2020-05-14:
The feature is now available with release 1.4. An [example program](https://github.com/MrYsLab/pymata4/blob/master/examples/wifi_blink.py) to blink the board LED on an ESP-8266 nodeMCU is provided as part of the release.

closed 2020-05-14 by MrYsLab

d-a-v, 2020-05-15:
I just tried and it works a treat !
I just added the IP/port/gpio to the servo example and the device started moving.
Thanks very much for such a fast support :)

MrYsLab, 2020-05-15:
My pleasure. FYI, I am now adding support for pymata-express, the Python asyncio sibling of pymata4.

ivanaquarianto, 2021-05-28:
Hello, I am new to Pymata4, my current goal is to control the arduino uno over ethernet using Python

I can't find a reference to be able to use Arduino Uno over Ethernet, is the use of Arduino over Ethernet already supported? or not?

MrYsLab, 2021-05-28:
Thanks for the question. I am assuming you mean the standard Uno that uses serial USB for programming and communication. Pymata4 uses the serial link and therefore would not provide ethernet connectivity.
However, you can use another one of my libraries, [python banyan](https://mryslab.github.io/python_banyan/), to do so. I would be happy to explain how this is done if you like. The documentation is rather detailed, but using banyan is really fairly simple. This would require another computer, such as a Raspberry Pi with WiFi capabilities. 

If you are using an Uno with an Ethernet interface, perhaps [StandardFirmataWifi](https://github.com/firmata/arduino/tree/master/examples/StandardFirmataWiFi) will work with pymata4.

I have tested pymata4 using StandardFirmataWiFi running on an ESP-8266, and that works.

To use pymata4 with wifi, you would [set the ip_address and ip_port values](https://mryslab.github.io/pymata4/init/) when it is instantiated.

Please let me know if you have any further questions.

ivanaquarianto, 2021-05-29:
Thank you very much for the answer, for now, I'm trying to use uno with W5100 ethernet shield,

I will try to use StandardFirmataWifi,
To use it, does the Arduino module have to be connected to a router / network?
Is there any arduino code that must be changed / configured first to be able to use the Ethernet Shield on StandardFirmataWifi?

Again, thanks for the answers, I'm still learning to use arduino and still have a lot to learn.

MrYsLab, 2021-05-29:
Since the W5100 is an ethernet interface and not Wifi, you will need to connect a cable between it and your router.

StandardFirmataWiFi will not work with this board, so you will need to use [StandardFirmataEthernet](https://github.com/firmata/arduino/tree/master/examples/StandardFirmataEthernet). A [configuration file](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataEthernet/ethernetConfig.h) must be configured before compiling and uploading to your Uno. You will need to go through all of the options in this file and set them accordingly. 

I know where this file is located in Linux but have no idea where it is in Windows. You will need to find it. It is not directly accessible from the Arduino IDE.

For step 2 in the file, comment out:
```
#define remote_ip IPAddress(10, 0, 0, 3)
```
This will establish the board as a server.

For step 3, you can leave the port as is.

For step 4, you must decide whether you want your router to assign an IP address using DHCP or if you wish to set up your router with a static address for the board. If you use DHCP, you will need to access the router to see what address was assigned.

For step 5, you need to enter the MAC address for your board.

Once you have done that, save the file and compile and upload the file to the Uno. Connect the ethernet cable between the router and the board. If you used DHCP, access the router and write down the IP address that was assigned.

For pymata4, when you instantiate the pymata4 class, you will need to specify the IP address and port and perhaps change the arduino_wait time as well.

So instead of writing:
```
board = pymata4.Pymata4()
```
You would write:
```
board = pymata4.Pymata4(arduino_wait=8, ip_address=192.168.111.111, ip_port=3030)
```
I modified the Arduino wait time to be 8 seconds instead of the standard 4 seconds in the example above. The IP address is one that I made up. Use the address that the router is using for the board. If you did not modify the port number in the config file, then use 3030.

Hopefully, when you start your python script, a connection will be successful, and you can test your code. If the connection is successful, you can try and remove the arduino_wait parameter and see if 4 seconds is enough time.

Please let me know if this works or not.

ivanaquarianto, 2021-05-31:
yes, it works

I have also tried removing the arduino_wait parameter and it still works perfectly on the servo code

Thank you!

MrYsLab, 2021-05-31:
That's great! Thanks for letting me know.