MrYsLab/pymata4 #6: Feature Request: including razor-9dof-ahrs as sensor

mockbot, 2020-05-08:
Any chance to include https://github.com/Razor-AHRS/razor-9dof-ahrs into special firmata software ?
https://github.com/Razor-AHRS/razor-9dof-ahrs

I'm not sure, if this fits into concept or performance

MrYsLab, 2020-05-08:
Thanks for the question. I apologize ahead of time - this is going to be rather lengthy.

Looking at the current Sparkfun 9dof offerings, each of the breakouts uses i2c as the hardware/software interface to the 9dof devices. Firmata, and therefore pymata4 provides an i2c interface. What would need to be done, is to translate the Arduino i2c calls embedded in the library code to Firmata i2c calls.

To see an example for the [MMA8452Q triple-axis accelerometer](https://www.sparkfun.com/products/13926) you can [look at this code](https://github.com/MrYsLab/pymata-aio/blob/master/examples/mma8452q3.py) that performs this translation.

As you can see, it is not a trivial task for a device as complex as the MMA84552Q.

For simpler devices, such as the [tmp102 ](https://www.sparkfun.com/products/13314), [the code](https://github.com/MrYsLab/PyMata/blob/master/examples/i2c/pymata_i2c_read/i2c_read_polled.py) is much simpler to implement.

The scope of Firmata, in general, is to provide an abstract view of the Arduino hardware. With the [number of registers](https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/src/MPU9250_RegisterMap.h) to control in some of the 9dof devices and the permutations and combinations that can be assembled to configure the devices, it really isn't practical to implement this as a hard-coded part of Firmata.

You could take a look at this [Sparkfun sketch](https://cdn.sparkfun.com/assets/6/3/8/e/5/RazorI2CSlaveExample.zip) to see how to implement controlling an IMU over serial link to see if it meets your needs, but it also is fairly complex if you need to modify things.

So to make this long story even longer, the answer to your question is, if you wish to program and utilize one of the 9dof Sparkfun boards with pymata4 (or any other Firmata client), you will need to write the Python code using the Firmata i2c primitives. I know this is not the answer you were hoping for, but in the scope of Firmata, it is the best I can give. If you wanted to try and customize pymta4 to do what you are asking, the first thing that you would need to do is to modify StandardFirmata or FimrataExpress to incorporate all of the i2c bit twiddling for the specific device in mind. You would need to add new [Firmata Sysex](https://github.com/firmata/protocol/blob/master/protocol.md#sysex-message-format) messages to expose an interface to the functionality, and finally, modify pymata4 to communicate these new Sysex messages across the serial link.

If you wanted to try and tackle this, I could provide guidance where possible.

MrYsLab, 2020-05-09:
I am closing this issue. You may still comment here and I will see it.

closed 2020-05-09 by MrYsLab