MrYsLab/pymata4 #7: arduino uno does not work

ouki-wang, 2020-05-10:
I use python3.7.4 and arduino uno burned standardfirmata using arduinoIDE1.8.12, and change the baudrate to 115200. and then I tried to test examples and found the error like this

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...
Traceback (most recent call last):
  File "digital_pin_output.py", line 16, in <module>
    board = pymata4.Pymata4(com_port="com4")
  File "C:\Python37\lib\site-packages\pymata4-1.2-py3.7.egg\pymata4\pymata4.py", line 204, in __init__
  File "C:\Python37\lib\site-packages\pymata4-1.2-py3.7.egg\pymata4\pymata4.py", line 368, in _manual_open
RuntimeError: Invalid Arduino ID reply lengtherror

MrYsLab, 2020-05-10:
If you want to use StandardFirmata, then you must set the baud rate to 57600. The error you are getting is correct since pymata4 cannot communicate with the Arduino and StandardFirmata at that rate. Please be aware, that the features that FirmataExpress provides, 115200 baud, an interface to HC-SR04 devices, support for a single stepper motor, control of a piezo buzzer is not possible with StandardFirmata. 

If you wish to try FirmataExpress, then you may find [installation instructions](https://mryslab.github.io/pymata4/firmata_express/) here.

I am closing this issue since the code is behaving as it should, but you may add a comment if you wish and I will see it.
Please let me know if you have any questions.

closed 2020-05-10 by MrYsLab

ouki-wang, 2020-05-11:
It has the same problem after I used baudrate 57600

ouki-wang, 2020-05-11:
I run digital_pin_output.py in jupyter, it works well, but when it stops running, a dialog box pops up, python process crash

MrYsLab, 2020-05-11:
Here is a screenshot of my run of the same script and I am not seeing any errors. Did you modify the script in any way?

[image]

ouki-wang, 2020-05-12:
yes，It work well。the problem will happen after I click  these two buttons alternately，just like this

MrYsLab, 2020-05-12:
I am unable to translate the error message because it is in the form of an image and not text, but most likely what you are seeing is an exception being thrown by  Jupyter in response to the program abruptly stopping when you hit the stop button.

I am not very familiar with how Jupyter notebooks work. If you run the program using a Python interpreter outside of Jupyter, meaning from a Command Window, and hit Control-C, the program will stop without any exceptions.

I apologize, but you will need to research how Jupyter handles exceptions and how you need to modify the code to avoid seeing the exception messages. If you are unfamiliar with exceptions, you may wish to read [this article.](https://realpython.com/python-exceptions/)

gide2578, 2020-10-21:
Try to remove the line sys.exit(0).