MrYsLab/pymata4 #9: Do you plan to support ws2812 light strips?

ouki-wang, 2020-05-21:
If so, when will it be supported?

MrYsLab, 2020-05-21:
Thanks for the question. I don't think that integrating ws2812 support into Firmata is likely to work.
The Firmata sketch has a loop time of about 19 ms and most likely, because of the nature of the loop, the refresh rate may not be consistent enough and you might see the LEDs blink intermittently.

You may experiment with adding ws2812 support to [FirmataExpress](https://github.com/MrYsLab/FirmataExpress) and if that is successful, I would be willing to incorporate this into both pymata4 and FirmataExpress. To verify if things would work or not, initially, you would not need to provide the code to process any Firmata commands but only demonstrate the ability to control a strip using static code being refreshed in the loop section of the sketch.

MrYsLab, 2020-05-21:
I am going to close this issue since I do not plan to provide support for neopixels. However, if you have any comments, you can leave them here and I will see them. Also, if you wish to attempt implementing this on your own, I will be happy to provide any guidance I can.

closed 2020-05-21 by MrYsLab