MrYsLab/pymata4 #10: I tested dht11 and got the following print

ouki-wang, 2020-05-22:
(1280.0, 691.6, 1590128317.5708947)
(1280.0, 691.7, 1590128318.5233834)
(1280.0, 691.6, 1590128319.8913765)

It's a wrong result

martynwheeler, 2020-05-22:
Have you tested the sensor using other software?

MrYsLab, 2020-05-22:
@martynwheeler The calculation in pymata4 for the dht11 may be incorrect compared to the [dhtnew library code. ](https://github.com/RobTillaart/Arduino/blob/65f3ec50f4e0ee853e9cbfabff80511306588395/libraries/DHTNEW/dhtnew.cpp#L95)

I think the pymata4 code may also have missed the bit for negative temperatures. I will check this out later today to verify. If changes are needed I will update the file.

martynwheeler, 2020-05-22:
Thanks Alan

MrYsLab, 2020-05-22:
@ouki-wang I have uploaded version 1.5 to pypi. To upgrade pymata4 on your computer if you are using Windows please type:
```
pip install pymata4 --upgrade
```

For Linux and mac type:
```
sudo pip3 install pymata4 --upgrade
```

I do not own a dht11, so please try the update and if it works, please close this issue. If it still giving incorrect output, please copy and paste the output here. 

Thanks.

MrYsLab, 2020-05-24:
I am closing this issue. If you are still having issues with the dht11, please leave a comment here and I will reopen the issue.

ouki-wang, 2020-05-25:
I have upgraded pymate4 library，bit it doesn't work

`Searching for an Arduino configured with an arduino_instance = 1
Arduino compatible device found and connected to COM4

Retrieving Arduino Firmware ID...
Arduino Firmware ID: 2.5 FirmataExpress.ino

Retrieving analog map...
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Python37\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Python37\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Python37\lib\site-packages\pymata4\pymata4.py", line 1857, in _reporter
    method(response_data)
  File "C:\Python37\lib\site-packages\pymata4\pymata4.py", line 1497, in _dht_read_response
    raise RuntimeError(f'Unknown DHT Sensor type reported: {data[2]}')
RuntimeError: Unknown DHT Sensor type reported: 56

poll pin 6: humidity=0 temp=0 time of last report: 2020-05-25 13:29:05
poll pin 6: humidity=0 temp=0 time of last report: 2020-05-25 13:29:05
poll pin 6: humidity=0 temp=0 time of last report: 2020-05-25 13:29:05
poll pin 6: humidity=0 temp=0 time of last report: 2020-05-25 13:29:05`

ouki-wang, 2020-05-25:
I connected dht11 to D6 and dht22 to D7，modified the sensor_type on D6 from 22 to 11，and got the followed print

Searching for an Arduino configured with an arduino_instance = 1
Arduino compatible device found and connected to COM4

Retrieving Arduino Firmware ID...
Arduino Firmware ID: 2.5 FirmataExpress.ino

Retrieving analog map...
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

Pin: 7 DHT Type: 22 Humidity:46.2, Temperature: 27.9 Timestamp: 2020-05-25 13:43:39
Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Python37\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Python37\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Python37\lib\site-packages\pymata4\pymata4.py", line 1857, in _reporter
    method(response_data)
  File "C:\Python37\lib\site-packages\pymata4\pymata4.py", line 1497, in _dht_read_response
    raise RuntimeError(f'Unknown DHT Sensor type reported: {data[2]}')
RuntimeError: Unknown DHT Sensor type reported: 53

poll pin 6: humidity=0 temp=0 time of last report: 2020-05-25 13:43:40
poll pin 7: humidity=46.2 temp=27.9 time of last report: 2020-05-25 13:43:39
poll pin 6: humidity=0 temp=0 time of last report: 2020-05-25 13:43:40
poll pin 7: humidity=46.2 temp=27.9 time of last report: 2020-05-25 13:43:39

MrYsLab, 2020-05-25:
Thanks for the input. I was able to recreate what you are seeing and have fixed the issue with version  1.6. It is available on pypi. Please update pymata4 again and retest. It should work with the correct values and not geneate an exception.

If you are still seeing errors, please let me know. Thanks.

ouki-wang, 2020-05-26:
I got the following print after upgade pymata4 to v1.6
dht11 give the wrong value

Opening all potential serial ports...
        COM4

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance = 1
Arduino compatible device found and connected to COM4

Retrieving Arduino Firmware ID...
Arduino Firmware ID: 2.5 FirmataExpress.ino

Retrieving analog map...
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

Pin: 7 DHT Type: 22 Humidity:55.0, Temperature: 25.8 Timestamp: 2020-05-26 16:9:45
Pin: 6 DHT Type: 11 Humidity:6.3, Temperature: 3.0 Timestamp: 2020-05-26 16:9:46
poll pin 6: humidity=6.3 temp=3.0 time of last report: 2020-05-26 16:9:48
poll pin 7: humidity=55.0 temp=25.8 time of last report: 2020-05-26 16:9:47
poll pin 6: humidity=6.3 temp=3.0 time of last report: 2020-05-26 16:9:52
poll pin 7: humidity=54.9 temp=25.8 time of last report: 2020-05-26 16:9:53
poll pin 6: humidity=6.3 temp=3.0 time of last report: 2020-05-26 16:9:58
poll pin 7: humidity=54.9 temp=25.8 time of last report: 2020-05-26 16:9:57

MrYsLab, 2020-05-26:
I have ordered dht11 sensors and should have them in 3 -6 days. I will test and report the results.

MrYsLab, 2020-05-29:
I have updated both pymata4 and FirmataExpress and tested them with DHT11 devices. The values look correct. Please make sure that you update FirmataExpress before using version 1.7 of pymata4.
Thanks for your patience.

I am closing this issue, but if you have any questions or you feel it needs to be reopened, please leave a comment here.

Thanks

closed 2020-05-29 by MrYsLab