MrYsLab/pymata4 #16: Support for HMC5883L

davizucon, 2020-06-09:
Hello,
   I'm trying to get work HMC5883L base on i2c_adxl345_accelerometer.py, but no luck.
   I made some tests/changes like:

```
    my_board.set_pin_mode_i2c()
    # set up power and control register
    my_board.i2c_write(30, [2])
    time.sleep(.1)
    my_board.i2c_write(30, [0])
    time.sleep(.1)

    # set up the data format register
    my_board.i2c_write(30, [3])
    time.sleep(.1)

    read_count = 20
    while True:
        # read 6 bytes from the data register
        my_board.i2c_read(address, 4, 6, the_callback)
```

using this code as example: [Arduino C Code example](http://www.electronoobs.com/eng_arduino_tut77.php)

Thanks

MrYsLab, 2020-06-09:
I am assuming you have run the code in the link above and it works. I cannot verify this since I do not own the device.

Here is what I believe the script should look like:

```
import sys
import time
from pymata4 import pymata4

# the call back function to print the 3 data registers
def the_callback(data):
    """
    :param data: [pin_type, Device address, device read register, x data pair, y data pair, z data pair]
    :return:
    """
    print(data)

def hmc(my_board):
    # device address = 30
    my_board.set_pin_mode_i2c()

    # set up mode register
    my_board.i2c_write(30, [2, 0])
    time.sleep(.1)

    # start the continuous reads
    my_board.i2c_read_continuous(30, 3, 6, the_callback)

    # do nothing and wait for the data
    while True:
        try:
            time.sleep(.1)
        except KeyboardInterrupt:
            my_board.shutdown()
            sys.exit(0)

board = pymata4.Pymata4()
try:
    hmc(board)
except KeyboardInterrupt:
    board.shutdown()
    sys.exit(0)
```

If this works, please close this issue.

davizucon, 2020-06-10:
Thank you for converting the code!!!  I'll keep it for future devices, but the make some research the device HMC5883L actually is QMC5883L.
Related with [HMC5883L Compass Module Communicating but all zero's for x,y,z](https://forum.arduino.cc/index.php?topic=482179.0)

I'm closing the issue, maybe in future, add support for QMC5883L

Best

closed 2020-06-10 by davizucon