MrYsLab/pymata4 #17: Support for STM32

mklomp, 2020-06-12:
Hi,

Trying to move from pymata to pymata4 I run into an issue with the STM32F103C8 (black pill). In the old situation we got pymata working with FirmataPlusDue uploaded to the STM32. After uploading FirmataExpress to the SMT32 and running pymata4 I get the error "Firmata Sketch Firmware Version Not Found". I get the same error when I upload StandardFirmata. 

After uploading FirmataPlusDue again pymata4 is able to connect, but is not able to get callback working and I get no values. Although not sure whether this is a pymata4 or FirmataExpress issue, I suppose it is the latter. So on my STM32 I have:

StandardFirmata + pymata4 -> Error
FirmataExpress + pymata4 -> Error
FirmataPlusDue + pymata4 -> Connecting, but not working
FirmataPlusDue + pymata -> Working (but old versions we want to ditch)

Given that SMT32 is in Boards.h makes me believe it is supposed to work without a special Due version like FirmataPlus (as far as I can see the only difference between those two is some libraries that are not compatible with AMR/Due/STM32). 

Hope you can help.

MrYsLab, 2020-06-12:
I agree that  FirmataExpress is not the cause, which leaves 2 variables, pymata4, and the black pill. 
Since things are working for your nano, that leaves the black pill since pymata4 has no knowledge of the board type it is communicating with, and other than creating a data structure to hold the information for all the pins on the board, does not discriminate between board types.

It appears that after getting the number of pins and firmware version numbers, communications is shutting down for some reason. Getting the initial information from FirmataExpress uses the Firmata protocol and since we are seeing that work correctly, know that the serial link, at least for a while, is working properly.

The only thing that I can see that is different between PyMata and pymata4, is that pymata4 sets the [FirmataExpress sample time](https://github.com/MrYsLab/pymata4/blob/3915970eb6fbb751d31f489af021c56d1aa0ff97/pymata4/pymata4.py#L326).

The easiest way to test if this is the cause is to load an earlier version of pymata4. To do that, if you are using Linux or macOS please use the following command:

```
sudo pip3 install pymata4==1.1 --upgrade
```
If you are using Windows:

```
pip3 install pymata4==1.1 --upgrade
```

This will roll pymata4 back to an earlier release. 

Please let me know if that solves the problem.

Thanks.

mklomp, 2020-06-12:
I rolled back to 1.1 but unfortunately same results. Let me know if you have any other suggestions. Would love to move to pymata4 (and get rid of python2).

MrYsLab, 2020-06-12:
I am a total loss at this point. I am wondering if all communication is lost after pymata4 checks the board for the number of pins and version.

Perhaps you can run [this example](https://github.com/MrYsLab/pymata4/blob/master/examples/retrieve_firmware_version.py).

What you should see on your console is this:

```
python3 retrieve_firmware_version.py
pymata4:  Version 1.8

Copyright (c) 2020 Alan Yorinks All Rights Reserved.

Opening all potential serial ports...
	/dev/ttyACM0

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance = 1
Arduino compatible device found and connected to /dev/ttyACM0

Retrieving Arduino Firmware ID...
Arduino Firmware ID: 1.1 FirmataExpress.ino

Retrieving analog map...
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

1.1 FirmataExpress.ino

```
If you don't see the last line, then the Black Pill for some reason has shutdown serial communication. If you do see the last line, then [run this example](https://github.com/MrYsLab/pymata4/blob/master/examples/analog_input_with_time_stamps.py), [but change this line ](https://github.com/MrYsLab/pymata4/blob/3915970eb6fbb751d31f489af021c56d1aa0ff97/examples/analog_input_with_time_stamps.py#L71) from:
```
    my_board.set_pin_mode_analog_input(pin, callback=the_callback, differential=5)
```
using your pin number to:

```
    my_board.set_pin_mode_analog_input(YOUR PIN NUMBER callback=the_callback, differential=0)
```

This will force a print of the value returned for each iteration and not suppress any small changes in values. Look at the TX Led and it should be lit continuously.

mklomp, 2020-06-12:
Ok, some update after a bit of very premature debugging.

When I run your first example, I indeed get back the Firmata version. Currently running StandardFirmata 2.5.8 (and still on pymata 1.1): 

```
pymata4:  Version 1.01

Copyright (c) 2020 Alan Yorinks All Rights Reserved.

Opening all potential serial ports...
	/dev/ttyACM0

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance = 1
Arduino compatible device found and connected to /dev/ttyACM0

Retrieving Arduino Firmware ID...
Arduino Firmware ID: 2.5 StandardFirmata.ino

Retrieving analog map...
Auto-discovery complete. Found 35 Digital Pins and 10 Analog Pins

2.5 StandardFirmata.ino
```

So that looks good. I already was running my test script with differential=0 to indeed show all updates. But, I am onto something. Current setup:

Blackpill with StandardFirmata 2.5.8. I've connected an IR sensor to analog pin 0 (A0 on the blackpill). Then tried to read out this pin with callback (and differential=0). This still gives no data as mentioned mentioned earlier. So now for the debugging: I checked if data comes in at [_analog_message](https://github.com/MrYsLab/pymata4/blob/3915970eb6fbb751d31f489af021c56d1aa0ff97/pymata4/pymata4.py#L1422) and it does. Only I got data: [5, 127, 7].

So, for some reason it thinks it is on pin5 (instead of 0, which I called in set_pin_mode_analog_input). To get it back to 0, I changed [this line](https://github.com/MrYsLab/pymata4/blob/3915970eb6fbb751d31f489af021c56d1aa0ff97/pymata4/pymata4.py#L1430) into:

`pin = data[0] - 5`

And put the cable onto A5 (instead of A0) and now it works. So something in the indexing of the pins is going wrong. Will look into it later, but any ideas/thoughts are more than welcome!

mklomp, 2020-06-12:
Just checked what I get back when prniting the analog map. For both PyMata and pymata4 I get: [127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 127, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 127, 127, 127, 127, 127]

Which to me seems ok.

MrYsLab, 2020-06-12:
Thanks for taking the time to debug.

I think the problem is in one of the board description files, either on the Firmata side (boards.h) or somewhere in the Arduino IDE files. First, pymata4 reports that there are 35 digital pins and 10 analog pins. The analog number is correct, but I think the digital number is wrong.

Also, typically a pin map shows all the digital pins followed by all the analog pins. I have never seen digital pins followed by analog pins followed by additional digital pins. 

The concept of Analog pin numbers (A0, A1, etc.) is an Arduino abstraction. What I mean is that on an Uno, for example, pin A0 is actually pin number 14. The way that [pymata4 determines the pin number to use](https://github.com/MrYsLab/pymata4/blob/3915970eb6fbb751d31f489af021c56d1aa0ff97/pymata4/pymata4.py#L317) for the first analog pin number is to take the total number of pins found and subtract the total number of analog pins. This is an Arduino standard method of mapping. PyMata does not use this method (which was a bug) and therefore worked.

The mapping algorithm is why you are seeing things off by 5.

You can print the supported pin modes for each pin by executing [this example.](https://github.com/MrYsLab/pymata4/blob/master/examples/retrieve_capability_report.py)

So the problem is definitely not in pymata4. You could try playing with hardcoding the number of pins in FirmataExpress boards.h for the stm boards, but since this may break things for other boards, I will not be making the changes on my end.

MrYsLab, 2020-06-13:
Thanks for taking the time to help debug the issue. Since the problem, I believe, is in the way that the stm32 folks defined the variants.h and boards.h files for the Black Pill, and not in either pymata4 or FirmataExpress, I am going to close this issue.

mklomp, 2020-06-17:
Thanks for your help. I now 'fixed' the offset for this board in the Boards.h and everything seems to work.

MrYsLab, 2020-06-17:
Thank you for letting me know it worked.

MrYsLab, 2020-07-23:
@mklomp I just ordered a black pill and hope to have it here by mid August. I just tried compiling FirmataExpress an

MrYsLab, 2020-07-23:
@mklomp I just ordered a black pill and should have it here in mid-August. I tried compiling for black pill by placing the following in the Arduino Preferences  selection:
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
[image]

I then added the boards using the board manager for this library.
[image]

If I compile for StandardFirmata or FirmataExpress I get a compile error. There was a change that the Arduino Firmata folks made that did not make it into the latest Arduino IDE. 

A change will need to be made in FirmataExpress as well, which I will test when I get the board. If you wish to try this yourself,
a line needs to be changed in [FirmataExpress.h.](https://github.com/MrYsLab/FirmataExpress/blob/1425a84ff145128454f192cf5eb6d093fe9f52ea/FirmataExpress.h#L140)

The original line was:
```
    friend void FirmataMarshaller::encodeByteStream (size_t bytec, uint8_t * bytev, size_t max_bytes = 0) const;
```
And the changed line should be:
```
    friend void FirmataMarshaller::encodeByteStream (size_t bytec, uint8_t * bytev, size_t max_bytes) const;
```
I will update this issue once I get the board. I do not know if any of this will work, but hopefully it will solve your issue.

mklomp, 2020-07-24:
Hi @MrYsLab. Good to hear that you ordere a black pill as well. The friend error was indeed already fixed in the StandardFirmata repo, but indeed not yet in the latest release. To get things working we forked FirmataExpress and made some fixes in there to get it to work:

[The friend error as you mentioned](https://github.com/zoef-robot/FirmataExpress/commit/47e5cce844811a3a9ef9b814bc52232cbde75a2e), also already in StandardFirmata repo
[We added the offset of 5, as mentioned above](https://github.com/zoef-robot/FirmataExpress/commit/9ed21d55c6701d1621c8ab0ed4311c2c5b4c26d5), the StandardFirmata repo has an updated Boards.h with another STM32 definition. Unfortunately that did not fix the issue. I do not know where the issue comes from, so did not spend time to create an issue.
[We needed to do det some pins to 0 which on the STM32 are 1 on boot, but is disputable](https://github.com/zoef-robot/FirmataExpress/commit/1ded947cd57ca0f8bd7d91deebdeaaf1e58a858d).

MrYsLab, 2020-07-24:
I don't think adding 5 is correct. If you look at the stm32duino variant code for the[ PILL 303](https://github.com/stm32duino/Arduino_Core_STM32/blob/17fd01f9ff569cc55de975da6a5742a8aab6d004/variants/PILL_F303XX/variant.h#L64), you will see that the Analog pins start at pin number 20. 
Also, when I compile on the Arduino IDE, it sets the board type to BLACKPILL_F303CC, which is not in boards.h.
I may have to add this as an additional type to boards.h for FirmataExpress.
Again, I will let you know what I determine once I get the board.

MrYsLab, 2020-08-03:
I received the board today, but it does not have a USB bootloader. I've ordered an ST-Link that should arrive within 5 days. Hopefully I will be able to program the beast and use the ST-Link for Firmata serial communications. What an adventure ;-).

MrYsLab, 2020-08-04:
@mkomp I have also ordered a serial to TTL board because there is no available Arduino bootloader for the board I have. This may mean I will need to modify FirmataExpress to user Serial1 for this board. I have also now ordered an STM32 Black Pill that either comes with an Arduino bootloader or if not, should be able to add one. I am assuming that your board has an Arduino boot loader and the board is recognized when you plug it into a USB port. Please let me know if this assumption is true or not. 

The second board will arrive at the end of August, but hopefully, I should be able to troubleshoot things with my current board.
Thanks for the patience.

mklomp, 2020-08-06:
Hi @MrYsLab, thanks for diving into this! I indeed have the board with Arduino bootloader on it. I just ordered it [soldered and with Arduino bootloader](https://robotdyn.com/stm32-arm-arduino-mini-system-dev-board-blue-pill-with-arduino-bootloader.html
). But indeed, you should also be able to flash the bootloader yourself as well.

It will be recognised when plugged in. I think you already came far regarding the building process. For uploading you need the right settings (as mentioned earlier, the CDCgen part). Let me know if you need more info, we have this board working (with the 5 offset 'fix') with the Arduino IDE.

MrYsLab, 2020-08-09:
I am not having much luck at this point. I was able to program my device with an ST-Link - both a simple blink program and with FirmataExpress. I finally received the USB to TTL converter, and I don't know what I did, but I can no longer program the board. I made sure to connect 3v to 3v. The ST-Link programmer no longer can detect the board. I am giving up with the current board.
My next board should be here sometime between now and the end of the month. Hopefully, the new board will work with a USB boot loader and without having to use all these add-ons. Let's see what happens when that board arrives.

MrYsLab, 2020-08-31:
I finally received the board. Unfortunately, I am able to communicate with FirmataExpress or StandardFirmata. I tried a simple test program to see if serial reads and writes work and they do, but for some reason, I am unable to receive any SysEx responses from Firmata. Have you seen this behavior with your boards? If yes, what was the fix? I am not sure if putting debug statements into FirmataExpress will help since it uses Sysex commands to send the data back. I tried adding delays after sending a Sysex command and sending the command multiple times, but neither work.

MrYsLab, 2020-08-31:
I just tried reinstalling FirmataExpress using the ST-Link and it looks like I bricked the board. At this point, I am sorry, but I am going to have to give up since I can't seem to make things work. Sorry, but these boards are just too flaky for me. I am going to close this issue. If things are working for you, then please continue doing what you are doing.

closed 2020-08-31 by MrYsLab

mklomp, 2020-09-10:
Thanks for all the time and effort you've put into it! I have only programmed it thought USB with the Arduino bootloader, so no clue why the TTL convertor version does not work. When uploading though USB we did not have any problems with not receiving SysEx responses. That part seemed to work fine. 

We did though manage to brick the board quite easily. This happened when not all settings in the Arduino IDE were correct and/or when there is still a bug in the code. At that  point the dfu part was unable to connect automatically so we had to manually press the reset button when flashing. That unbricked it and was only needed I our earlier days when we still tried to get everything working.

Thanks again for all your effort!