MrYsLab/pymata4 #19: Possibly documentation error for pwm_write?

dshaw619, 2020-06-26:
The docstring for pwm_write show on https://mryslab.github.io/pymata4/device_writes/ says ":param value: Pin value (0 - 0x4000)", but then the Notes say "The value parameter is typically set between 0 and 255" which would be 0 - 0xff wouldn't it?

MrYsLab, 2020-06-27:
Thanks for your question.

The comment is actually correct. Typically, one uses a value between 0 and 255, but the Arduino low-level library has a little known feature to control the PWM clock rate by setting the upper order bits. For some reason, the original authors of StandardFirmata decided to support this feature. I have never used it, nor have I seen it used.

The feature is documented on the Arduino site [here](https://www.arduino.cc/en/Tutorial/SecretsOfArduinoPWM). I find this documentation somewhat confusing and found that [this article ](https://www.arduinoslovakia.eu/blog/2017/7/16-bitove-rozlisenie-pwm-pre-arduino?lang=en)is easier to understand. The Firmata implementation supports 14 bits. I am not sure why not the full 16, but that is the reason for value 0x4000.

I am closing the issue, but if you have any additional comments or questions, you may post them here and I will see them.

closed 2020-06-27 by MrYsLab

dshaw619, 2020-06-27:
Thanks for the explanation.

Doug