MrYsLab/pymata4 #20: i can't use the analog_read  in samd21

wodenageshen, 2020-06-29:
hello;
         thanks for your software ,it's great.
        i download the firmataExpress.ino to arduino uno , it could work well (digital and analog_read) , but wen i use 
         samd21(seeeduino zero)  , the analog_read function could not work well ,  the return's is always (0,0).

MrYsLab, 2020-06-29:
I don't believe that StandardFirmata supports this board. To prove this, could you please do the following:
1. Install StandardFirmata on your seeeduino zero.
2. Modify [this line of the  analog input example](https://github.com/MrYsLab/pymata4/blob/aeab6607bfc98f8ad0c36adaa0ce27fa18d66264/examples/analog_input_with_time_stamps.py#L89)  from
```
board = pymata4.Pymata4()
```
to
```
board = pymata4.Pymata4(baud_rate=57600)
```
Then run the script to see if it works. If it does not then we can create an issue against StandardFirmata.

Please let me know the results.

wodenageshen, 2020-06-29:
Hi:

         Here is my test!

 

thanks

 

发件人: noreply@github.com [mailto:noreply@github.com] 代表 Alan Yorinks
发送时间: 2020年6月29日 20:18
收件人: MrYsLab/pymata4
抄送: wodenageshen; Author
主题: Re: [MrYsLab/pymata4] i can't use the analog_read in samd21 (#20)

 

I don't believe that StandardFirmata supports this board. To prove this, could you please do the following:

1.	Install StandardFirmata on your seeeduino zero.
2.	Modify this line of the analog input example <https://github.com/MrYsLab/pymata4/blob/aeab6607bfc98f8ad0c36adaa0ce27fa18d66264/examples/analog_input_with_time_stamps.py#L89>  from

board = pymata4.Pymata4()

to

board = pymata4.Pymata4(baud_rate=57600)

Then run the script to see if it works. If it does not then we can create an issue against StandardFirmata.

Please let me know the results.

—

#!/usr/bin/env python3
from pyfirmata import Arduino,util
import time
import sys
 
board = Arduino('COM28') # Change to your port
it = util.Iterator(board)
it.start()
board.analog[0].enable_reporting()
print ("Read Start")
for i in range(10):
    time.sleep(1)
    print(board.analog[0].read())
 
board.analog[0].disable_reporting()

"""

 Copyright (c) 2020 Alan Yorinks All rights reserved.

 This program is free software; you can redistribute it and/or

 modify it under the terms of the GNU AFFERO GENERAL PUBLIC LICENSE

 Version 3 as published by the Free Software Foundation; either

 or (at your option) any later version.

 This library is distributed in the hope that it will be useful,

 but WITHOUT ANY WARRANTY; without even the implied warranty of

 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU

 General Public License for more details.

 You should have received a copy of the GNU AFFERO GENERAL PUBLIC LICENSE

 along with this library; if not, write to the Free Software

 Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA

"""

import sys

import time

from pymata4 import pymata4

"""

This file demonstrates analog input using both callbacks and

polling. Time stamps are provided in both "cooked" and raw form

"""

# Setup a pin for analog input and monitor its changes

ANALOG_PIN = 2  # arduino pin number

POLL_TIME = 5  # number of seconds between polls

# Callback data indices

CB_PIN_MODE = 0

CB_PIN = 1

CB_VALUE = 2

CB_TIME = 3

#

def the_callback(data):

    """

    A callback function to report data changes.

    :param data: [pin_mode, pin, current_reported_value,  timestamp]

    """

    formatted_time = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(data[CB_TIME]))

    print(f'Analog Call Input Callback: pin={data[CB_PIN]}, '

          f'Value={data[CB_VALUE]} Time={formatted_time} '

          f'(Raw Time={data[CB_TIME]})')

def analog_in(my_board, pin):

    """

    This function establishes the pin as an

    analog input. Any changes on this pin will

    be reported through the call back function.

    Every 5 seconds the last value and time stamp is polled

    and printed.

    Also, the differential parameter is being used.

    The callback will only be called when there is

    difference of 5 or more between the current and

    last value reported.

    :param my_board: a pymata4 instance

    :param pin: Arduino pin number

    """

    my_board.set_pin_mode_analog_input(pin, callback=the_callback, differential=5)

    # run forever waiting for input changes

    try:

        while True:

            time.sleep(POLL_TIME)

            # retrieve both the value and time stamp with each poll

            value, time_stamp = board.analog_read(pin)

            # format the time stamp

            formatted_time = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(time_stamp))

            print(

                f'Reading latest analog input data for pin {pin} = {value} change received on {formatted_time} '

                f'(raw_time: {time_stamp})')

    except KeyboardInterrupt:

        my_board.shutdown()

        sys.exit(0)

# instantiate pymata4

board = board = pymata4.Pymata4(baud_rate=57600)

analog_in(board, ANALOG_PIN)

MrYsLab, 2020-06-29:
If you adjust the script with the pin number you are using, does this script run correctly. It was not clear from your post above.
Thanks.

wodenageshen, 2020-06-30:
It just could use pin  0-5  and all of these pins return 0 .  I had test arduino zero  and the result is the same as seeeduino zero .

 

thanks

 

 

 

 

发件人: noreply@github.com [mailto:noreply@github.com] 代表 Alan Yorinks
发送时间: 2020年6月29日 21:13
收件人: MrYsLab/pymata4
抄送: wodenageshen; Author
主题: Re: [MrYsLab/pymata4] i can't use the analog_read in samd21 (#20)

 

If you adjust the script with the pin number you are using, does this script run correctly. It was not clear from your post above.
Thanks.

MrYsLab, 2020-06-30:
Have you tried loading standardfirmata onto the zero? Does that work?

wodenageshen, 2020-06-30:
If I use arduino zero  ,it could not get the firmware id

 

 

发件人: noreply@github.com [mailto:noreply@github.com] 代表 Alan Yorinks
发送时间: 2020年6月30日 10:12
收件人: MrYsLab/pymata4
抄送: wodenageshen; Author
主题: Re: [MrYsLab/pymata4] i can't use the analog_read in samd21 (#20)

 

Have you tried loading standardfirmata onto the zero? Does that work?

MrYsLab, 2020-06-30:
Please answer my question. Does the zero work with the StandardFirmata sketch installed? Make sure you set the baud_rate as explained above for StandarFirmata.

wodenageshen, 2020-06-30:
No ，arduino zero or seeeduino zero  no matter installed standardfirmata sketch or firmataexpress sketch both could not work well whitch pymata4. The follow picture is on arduino zero.

 

 

发件人: noreply@github.com [mailto:noreply@github.com] 代表 Alan Yorinks
发送时间: 2020年6月30日 11:09
收件人: MrYsLab/pymata4
抄送: wodenageshen; Author
主题: Re: [MrYsLab/pymata4] i can't use the analog_read in samd21 (#20)

 

Please answer my question. Does the zero work with the StandardFirmata sketch installed? Make sure you set the baud_rate as explained above for StandarFirmata.

MrYsLab, 2020-06-30:
It appears that StandardFirmata does not support the Zero. Since FirmataExpress is built on StandardFirmata, it also does not support the Zero. This is documented on the Arduino Firmata GitHub pages in this[ issue.](https://github.com/firmata/arduino/pull/227)

I am sorry that until Firmata supports the Zero, I too cannot support it. This is not a pymata4 issue, but a Firmata issue.

closed 2020-06-30 by MrYsLab

MrYsLab, 2020-07-01:
@wodenageshen I just ordered a Seeed Wio Lite W600 which is supposed to be Arduino zero compatible. I should receive it sometime next week. I will let you know if it works for me or not, or if I see what you reported.

wodenageshen, 2020-07-02:
Thank you very much , I am looking forwared to using the new version.

 

发件人: noreply@github.com [mailto:noreply@github.com] 代表 Alan Yorinks
发送时间: 2020年7月2日 6:18
收件人: MrYsLab/pymata4
抄送: wodenageshen; Mention
主题: Re: [MrYsLab/pymata4] i can't use the analog_read in samd21 (#20)

 

@wodenageshen <https://github.com/wodenageshen>  I just ordered a Seeed Wio Lite W600 which is supposed to be Arduino zero compatible. I should receive it sometime next week. I will let you know if it works for me or not, or if I see what you reported.

MrYsLab, 2020-07-09:
I have bad news in that I received the Seeed Studio board and for some reason, the board uses a USB-C male connector. I cannot find a cable that will connect with this. iPhone cables use a male USB-C connector and therefore cannot be used - male to male does not work. As a result, I am returning the board. I also cannot find some other reasonably priced samd21 board, so I am sorry, but I cannot pursue this any further.