MrYsLab/pymata4 #21: RuntimeError: Invalid Arduino ID reply length

ipsod, 2020-09-22:
Hi.  I've got a board that I've uploaded StandardFirmata onto (Firmata library version 2.5.8, from Arduino IDE).

### Board and Firmata tested as working
I've tested that it works with the firmata_test program on a Windows VM (couldn't get firmata_test to run on my Linux machine because I couldn't find a library it required).

### initialization in PyMata4 fails
On Linux, when I try to initialize the board with `ard = pymata4.Pymata4('/dev/ttyUSB0')`, I get the error: "RuntimeError: Invalid Arduino ID reply length".  To try and see what was going on, I modified the PyMata source to output the variable which should contain the Arduino ID, `i_am_here`, along with the error message:
try 1:
`RuntimeError: ('Invalid Arduino ID reply length', [134, 24, 102, 0, 134, 134, 0, 102, 30, 102, 96, 6, 6, 6, 248, 158, 0, 96, 6, 6, 6, 24, 6, 96, 6, 120, 6, 134, 158, 0, 24, 6, 230, 158, 0, 6, 6, 96, 6, 6, 6, 248, 134, 0, 134, 158, 0, 248, 158, 0, 254, 158, 0, 126])`
try 2:
`RuntimeError: ('Invalid Arduino ID reply length', [134, 24, 102, 0, 134, 134, 0, 102, 30, 102, 96, 6, 6, 6, 248, 158, 0, 96, 6, 6, 6, 24, 6, 96, 6, 120, 6, 134, 158, 0, 24, 6, 230, 158, 0, 6, 6, 96, 6, 6, 6, 248, 134, 0, 134, 158, 0, 248, 158, 0, 254, 158, 0, 126])`

### system info
OS: Linux Mint 19.1 Cinnamon
Python 3.8.0
pymata4==1.10
arduino uno clone

MrYsLab, 2020-09-22:
To use StandardFirmata (without any modifications), the baudrate must be specified when instantiating Pymata4:

```
board = pymata4.Pymata4(baud_rate=57600)
```
Please let me know if that solves your issue.

ipsod, 2020-09-22:
That fixed it.  Thank you!

closed 2020-09-22 by ipsod