MrYsLab/pymata4 #24: Serial Comms Arduino

whiteteepoison, 2020-11-17:
Hello, I've tried looking for some documentation regarding communications over the RX and TX pins through pymata4, but could not find any. I am just looking to send & receive data by arduino using a simplex MAX485  setup (rs485, modbus RTU), then back to my computer running a slave server. 

Is this supported?

MrYsLab, 2020-11-17:
Thanks for the question. There are 2 articles that may help. 

The [first article ](https://www.instructables.com/Going-Beyond-StandardFirmata-Adding-New-Device-Sup/)is based on an older version of StandardFirmata, which had a simpler directory structure, and a [newer article](https://www.instructables.com/Going-Beyond-StandardFirmata-Revisited/) based on the latest directory structure. In the second article, I did not use the DHT library directly. I would suggest that you do not use this technique but use your MAX485 library directly.
Please keep in mind is that whatever library you decide to use, make sure that it does not block the loop portion of the Firmata sketch. 

Also, I have an alternative to using Firmata, called [Telemetrix](https://github.com/MrYsLab/Telemetrix4Arduino), that is simpler to extend than Pymata4 and FirmataExpress.
Data across the serial link for [Telemetrix](https://github.com/MrYsLab/telemetrix) uses an 8 bit byte as opposed to Firmata's 7 bit byte. In the user's guide, I have [a section ](https://mryslab.github.io/telemetrix/dht/)that describes how to extend both telemetrix (the Python client) and Telemetrix4Arduino (the Arduino server). There is also a Python [asyncio version](https://github.com/MrYsLab/telemetrix-aio) of the client.

If you have any questions, please feel free to either contact me here, or send an email to the address [on this page](https://github.com/MrYsLab) (under my avatar).

I am going to close this issue, but I will still see any comments you may make here. Also, if necessary, it can be reopened.

closed 2020-11-17 by MrYsLab

whiteteepoison, 2020-11-18:
Thanks for your help! I will follow your suggestion, Telemetrix seems to be the best way forward for my project.