MrYsLab/pymata4 #26: Communication between ST32 MCU over Serial Comms and Firmata not working.

Kaladin-Syl-WR, 2021-01-28:
Hello everyone. This is my first question on a public forum so please excuse any errors pertaining to the issue format, or if to little information is given.

I am currently doing a project for vacation work where I have to use Firmata on different STM32 Microcontrollers. I am using the Arduino IDE to program the STM32L476RG Nucleo with StandardFirmata. I use Pymata4 to communicate/interface with the MCU. So far everything works as expected when I use the MCU's normal USB connector for communicating with the MCU. For my task, next task I cannot use the USB connector to communicate with the MCU. So what I am currently trying to do is communicate with the MCU through the USART1 RX an TX pins using the FT232R USB to serial UART interface.

When trying to establish a connection with the MCU through the USB to serial interface. The python code gives the following error:
[image]

I am just using the following code blinky example to test the connection:
```
import sys
import time

from pymata4 import pymata4

board = pymata4.Pymata4(com_port="COM7", baud_rate=57600)
pin = 13

board.set_pin_mode_digital_output(pin)
board.digital_write(pin, 1)

# toggle the pin 4 times and exit
for x in range(4):
    print('ON')
    board.digital_write(pin, 1)
    time.sleep(1)
    print('OFF')
    board.digital_write(pin, 0)
    time.sleep(1)

board.shutdown()
```

I was able to confirm that the Arduino Uno successfully communicated with the MCU through the USB to serial interface with digital pins 0 and 1 as RX and TX respectively.

I have also confirmed that the STM32 MCU's USART1 does work over the USB to serial. I ran the Arduino MultiSerial example on the STM32 and was able read and write data through the USART1 pins/USB to serial interface.

So the problem seems to be between Pymata4 and the STM32's USART1, as it does work with the Arduino. 

I also did change the StandardFirmata serial instantiation to the correct serial interface for the STM32:
```
// STM32 serial ports/interface added
HardwareSerial Serial1 (USART1);

void setup()
{
  Firmata.setFirmwareVersion(FIRMATA_FIRMWARE_MAJOR_VERSION, FIRMATA_FIRMWARE_MINOR_VERSION);

  Firmata.attach(ANALOG_MESSAGE, analogWriteCallback);
  Firmata.attach(DIGITAL_MESSAGE, digitalWriteCallback);
  Firmata.attach(REPORT_ANALOG, reportAnalogCallback);
  Firmata.attach(REPORT_DIGITAL, reportDigitalCallback);
  Firmata.attach(SET_PIN_MODE, setPinModeCallback);
  Firmata.attach(SET_DIGITAL_PIN_VALUE, setPinValueCallback);
  Firmata.attach(START_SYSEX, sysexCallback);
  Firmata.attach(SYSTEM_RESET, systemResetCallback);

  // to use a port other than Serial, such as Serial1 on an Arduino Leonardo or Mega,
  // Call begin(baud) on the alternate serial port and pass it to Firmata to begin like this:

  // Serial for STM32
  Serial1.begin(57600);
  Firmata.begin(Serial1);
  
  // However do not do this if you are using SERIAL_MESSAGE1

  // Serial for Arduino
  // Firmata.begin(57600);
  while (!Serial1) {
    ; // wait for serial port to connect. Needed for ATmega32u4-based boards and Arduino 101
  }

  systemResetCallback();  // reset to default config
}
```

Hope that is enough information for you to help me. 
Thanks in advance.

MrYsLab, 2021-01-28:
@Kaladin-Syl-WR Thanks for the question. Using Firmata with the STM boards has been a major problem and prompted me to abandon Firmata and to create another library, similar to Firmata, but simpler to use. The name of the library is [Telemetrix](https://mryslab.github.io/telemetrix/). 
Like Firmata, you need to load a sketch onto the STM. The name of the sketch is Telemetrix4Arduino. It works for the STM as well as Arduino boards.

[image]

I have used this with a [Black Pill board](https://smile.amazon.com/CANADUINO-STM32-Black-STM32F411CEU6-128M-Extra/dp/B08946CFLZ/ref=sr_1_3?dchild=1&keywords=stm+black+pill&qid=1611839678&sr=8-3). This board does not include a serial loader, so I used an ST-Link to program it. I believe the Nucleo you are using includes an ST-Link and that should work as well. 

I am not sure if this article is accurate, since I don't own a Nucleo, but [here are some instructions](http://www.emcu.eu/2017/03/13/how-to-use-stm32-and-arduino-ide/) on using the Arduino IDE to program the Nucleo.

The [Python API ](http://htmlpreview.github.io/?https://github.com/MrYsLab/telemetrix/blob/master/html/telemetrix/index.html)is similar to Python4, but a somewhat simpler. 

Hopefully this helps, but if you have any questions, please feel free to ask. I am closing this issue, but if you post a reply I will still see it. If you are having issues with Telemetrix or Telemetrix4Arduino (the sketch), please post the issue at this [link](https://github.com/MrYsLab/telemetrix/issues).

closed 2021-01-28 by MrYsLab

Kaladin-Syl-WR, 2021-02-01:
Thank you very much for the clear and concise answer. I will definitely try that out and see if it solves my problem.