MrYsLab/pymata4 #27: Firmware Sketch Firmata Version not Found

thomasarmstrong98, 2021-01-31:
Hey, using the latest update of pymata4 and having an issue with error: "Firmata Sketch Firmware Version Not Found". I checked the docs and saw the troubleshooting steps although I am not sure what the problem is exactly. I am trying to retrieve data from an I2C temperature sensor on an Arduino. I can run tasks on the arduino using pyfirmata - so I know that the Firmata sketch is installed correctly but it is not picking it up when using pymata4. It is identifying the board but that's about it.

Wondering what steps I can take to fix or troubleshoot this further? Thanks.

MrYsLab, 2021-01-31:
If you are using pyfirmata then I am assuming you are using StandardFirmata. I recommend using [FirmataExpress](https://mryslab.github.io/pymata4/firmata_express/) which runs at twice the link speed of StandardFirmata. However, if you wish to still use StandardFirmata, you will need to [manually set the baud rate](https://mryslab.github.io/pymata4/init/#baud_rate) to 57600 when you instantiate pymata4.

Please let me know if you are able to resolve your issue.

thomasarmstrong98, 2021-01-31:
Thank you, I worked this out eventually around 5 minutes before your response! First time working with microcontrollers, so everything is a bit new - thanks for this great package!

closed 2021-01-31 by thomasarmstrong98

MrYsLab, 2021-01-31:
@thomasarmstrong98 I am glad it's working. If you need to extend the sketch to support devices that are not i2c, you might want to check out [Telemetrix](https://mryslab.github.io/telemetrix/). It is similar to Firmata but simpler to extend. 
The Telemetrix client commonly supports Arduino boards via serial, the ESP-8266 via WiFi, and STM32 boards via serial. There are separate sketches for each of the board types. Again thanks for letting me know and closing the issue.