MrYsLab/pymata4 #29: Support for PCA9685 Servo Driver

aahmad3, 2021-03-22:
Hello,

Does pymata4 currently support pca9685 servo driver? I tried running servos using pca9685 but to no avail. How can I proceed with adding said support? Any help is appreciated.

MrYsLab, 2021-03-22:
It can support the device since it is an i2c device, but the programming of the device would be part of your application. The i2c support  pymata4 provides is similar to that of the Arduino wire library. Essentially, it provides methods to send a variable number of bytes to an i2c device and to also read a variable number of bytes from the device. The library has no knowledge of the device you are using or how to interpret its registers. Again, that would be part of your application. 

To see what I mean, I provided an [example to control an ADXL345 accelerometer. ](https://github.com/MrYsLab/pymata4/blob/master/examples/i2c_adxl345_accelerometer.py) This device is very different from the pca9685, so you cannot use any of it for the pac9685, but I point to it because it illustrates how to read and write a device's registers.

I have not used this device, but perhaps [this library](https://github.com/Seeed-Studio/Seeed_PCA9685) can give you an idea of how to interact with pca9685 registers. 

What you would need to do is to port the register manipulations to pymata4 i2c calls.

If you are familiar with programming i2c devices, then it should be straightforward, but if you are not, it may be a bit of a challenge. 

Please let me know if I can answer any specific questions.

aahmad3, 2021-03-22:
Thank you for the quick response!

aahmad3, 2021-03-22:
I have a specific question. The registers on pca9685 that control servos are 8 bit long. pymata4 i2c_write works with 7 bits. So for example if I have to write 0x99 to a register on pca9685 I can write the first 7 bit byte as 0x19 leaving out the 1 bit MSB. But how can I write the 1 into the MSB of pca9685's register to make it 0x99, since pymata4 write command only sends a 7 bit byte?

Sorry if the answer is obvious but I can't seem to find it.

MrYsLab, 2021-03-22:
No need to apologize. Firmata is rather confusing.

You don't have to worry about the 7-bit vs 8-bit nonsense, that is handled all within pymat4 for you. For example in [i2c_wrte](https://github.com/MrYsLab/pymata4/blob/ede037df3f0d15f59cf7ad13fe6996f56a33139c/pymata4/pymata4.py#L856), all the shifting is within the method. This is true for when you do an i2c read as well. You just get back data in normal 8-bit format.

The only time you would need to deal directly with 7-bit bytes is if you are trying to extend the sketch on the Arduino. 

By the way, I developed a replacement for Firmata called [Telemetrix.](https://mryslab.github.io/telemetrix/). No 7-bit bytes or having to deal with the boards.h file. This version of Telemetrix works with the Arduino boards and STM32 boards as well.

I have a version for the ESP8266 using WiFi.

I am currently working on a version for the Raspberry Pi Pico.

If you have any other questions, I would be happy to answer.

aahmad3, 2021-03-22:
Thank you! I will check it out.

closed 2021-03-22 by aahmad3