MrYsLab/pymata4 #30: Analog Read (LDR Sensor) & DHT22 Read

Bingsitfx, 2021-04-10:
Hello 
Does pymata4 support LDR and DHT read simultaneously? i already try it but no luck

MrYsLab, 2021-04-10:
Could you please provide a simple Python script that exhibits this behavior so that I may reproduce it here. Also, can you answer the following questions:
1. If you operate each of the sensors individually do they function properly?
2. When combined do they both not work or just one?
3. If only  one is working, which sensor is it?

Bingsitfx, 2021-04-10:
I want to make a system based on fuzzy logic that can control temp, humidity & light intensity. 
1. Yes, both can run individually, no errors occurred.
2. If they combined, only analog read is working. DHT isnt working..

MrYsLab, 2021-04-10:
Could you please provide the test script you are using? Also, you might try [Telemetrix ](https://mryslab.github.io/telemetrix/) which is a replacement for pymata4.

>

martynwheeler, 2021-04-10:
I have a DHT22 and ldr working with pymata4 - I will dig out my script.

Bingsitfx, 2021-04-10:
This is my DHT code with Fuzzy logic [https://pastebin.com/31vy7T3P](url)
and this is my LDR code [https://pastebin.com/0ehsKNbE](url)

Bingsitfx, 2021-04-10:
> I have a DHT22 and ldr working with pymata4 - I will dig out my script.
> […](#)

Okay l ' m waiting for ya

martynwheeler, 2021-04-10:
Okay here is my code, it does not do anything fancy, just gets the ldr and dht readings, but it definitely works

```
#!/usr/bin/python3
# Import Libraries
import time
import requests
from pymata4 import pymata4
import os.path
import logging
from datetime import datetime
import paho.mqtt.client as mqtt
import json

# global variables
dht_pin = 2
dht_type = 22
analog_pin = 0
serial_port = '/dev/ttyAMA0' # raspberrypi v1
basedir = '/home/pi/python/SensorsDatabase/' # set basedir
thingspeak_key = '************************' # setup Thingspeak

def dir_exists(dirname):
    """
    checks if directory exists
    :param dirname:
    :return: bool
    """
    if os.path.isdir(dirname):
        return True
    else:
        logging.warning(' '.join(['The directory', dirname, 'does not exist']))
        return False

def initialise_sensors():
    """ create a connection to sensors via pymata4
    :return: Connection object or None
    """
    board = None
    try:
        # instantiate pymata4
        board = pymata4.Pymata4(com_port = serial_port, arduino_wait = 2)
        board.set_pin_mode_dht(dht_pin, sensor_type=dht_type)
        board.set_pin_mode_analog_input(analog_pin)
    except:
        logging.warning('Failed to connect to arduino')
    return board

def get_data(board, timeout = 10):
    """ get data from the arduino for the initialised sensors
    :param board: arduino connection object
    :param timeout: maximum duration to wait for response
    :return: sensor data in a dictionary
    """
    # define no data condition
    readings = (0, 0)
    # get current time
    start_time = time.time()
    # loop while trying to get a valid reading or until time reached
    while readings[0] == 0 or readings[1] == 0:
        if time.time() - start_time > timeout:
            print("Failed to get a reading")
            return
        else:
            dht = board.dht_read(dht_pin)
            readings = (dht[0], dht[1])
    # dht sensors successfully read, now grab light level and convert to %
    light = 100 * float(board.analog_read(analog_pin)[0]) / 1024
    return {'humidity': readings[0], 'temperature': readings[1], 'light': round(light, 1)}

def sendtoThingspeak(thingspeak_key, data):
    # send to thingspeak
    r = requests.post('https://api.thingspeak.com/update.json', data = {'api_key':thingspeak_key,
                                                                        'field1':data.get('temperature', ''),
                                                                        'field2':data.get('humidity', ''), 
                                                                        'field3':data.get('light', '')
                                                                        })

def publishData(data):
    client = mqtt.Client()
    client.connect("192.168.86.22", 1883, 60)
    client.publish("home/office/arduino/1", json.dumps(data));
    client.disconnect();

def main():
    if dir_exists(basedir):
        # start the logger
        logging.basicConfig(filename = ''.join([basedir, 'logfile.log']), level = logging.WARNING, format='%(levelname)s:%(asctime)s Function-%(funcName)s: %(message)s', datefmt='%d-%m-%Y %H:%M:%S')
        try:
            # connect to board and grab the data
            board = initialise_sensors()
            data = get_data(board)
            # send data to the cloud
            sendtoThingspeak(thingspeak_key, data)
            # mqtt
            #print(data)
            publishData(data)
        finally:
            board.shutdown()

if __name__ == '__main__':
    main()
```

MrYsLab, 2021-04-10:
@martynwheeler Thanks for providing that.
@Bingsitfx I just want to verify that on your Arduino you have [loaded FirmataExpress](https://mryslab.github.io/pymata4/firmata_express/). Although pymata4 is able to interact with StandardFirmata, DHT is part of the extended functionality that FirmataExrpess provides.

MrYsLab, 2021-04-10:
@Bingsitfx I forgot to mention that neither of the links you provided is pointing to your code. Here is the URL that both links are using: https://github.com/MrYsLab/pymata4/issues/url

Bingsitfx, 2021-04-10:
im so sory about that, I've edited it. 
My arduino already loaded with FirmataExpress.

MrYsLab, 2021-04-10:
I just ran this modified version of the DHT example and it is working for me. Output is below the source code.
```

import time
import sys
from pymata4 import pymata4

"""
Setup a pin for dht mode
One pin is set for a dht22 and another for dht11
Both polling and callback are being used in this example.
"""

#
POLL_TIME = 5  # number of seconds between polls

# Callback data indices
CB_PIN_MODE = 0
CB_PIN = 1
CB_VALUE = 2
CB_TIME = 3

def the_callback(data):
    """
    A callback function to report data changes.
    This will print the pin number, its reported value and
    the date and time when the change occurred

    :param data: [pin, current reported value, pin_mode, timestamp]
    """

    tlist = time.localtime(data[5])
    ftime = f'{tlist.tm_year}-{tlist.tm_mon:02}-{tlist.tm_mday:02} ' \
            f'{tlist.tm_hour:02}:{tlist.tm_min:0}:{tlist.tm_sec:02}'

    print(f'Pin: {data[1]} DHT Type: {data[2]} Humidity:{data[3]}, '
          f'Temperature: {data[4]} Timestamp: {ftime}')

def the_analog_callback(data):
    formatted_time = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(data[CB_TIME]))
    print(f'Analog Call Input Callback: pin={data[CB_PIN]}, '
          f'Value={data[CB_VALUE]} Time={formatted_time} '
          f'(Raw Time={data[CB_TIME]})')

def dht(my_board, callback=None):
    """
     This function establishes the pin as a
     digital input. Any changes on this pin will
     be reported through the call back function.

     :param my_board: a pymata4 instance
     :param callback: callback funtion
     """

    # set the pin mode - for pin 6 differential is set explicitly
    my_board.set_pin_mode_dht(6, sensor_type=22, differential=.01, callback=callback)
    my_board.set_pin_mode_analog_input(2, differential=5, callback=the_analog_callback)

    while True:
        try:
            time.sleep(POLL_TIME)

            # poll the first DHT
            value = board.dht_read(6)

            # format the time string and then print the data
            tlist = time.localtime(value[2])
            ftime = f'{tlist.tm_year}-{tlist.tm_mon:02}-{tlist.tm_mday:02} ' \
                    f'{tlist.tm_hour:02}:{tlist.tm_min:0}:{tlist.tm_sec:02}'
            print(f'poll pin 6: humidity={value[0]} temp={value[1]} '
                  f'time of last report: {ftime}')
        except KeyboardInterrupt:
            board.shutdown()
            sys.exit(0)

board = pymata4.Pymata4()

try:
    dht(board, the_callback)
except KeyboardInterrupt:
    board.shutdown()
    sys.exit(0)

```

And I got the following output:
```
pymata4:  Version 1.10

Copyright (c) 2020 Alan Yorinks All Rights Reserved.

Opening all potential serial ports...
	/dev/ttyACM2

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance = 1
Arduino compatible device found and connected to /dev/ttyACM2

Retrieving Arduino Firmware ID...
Arduino Firmware ID: 1.1 FirmataExpress.ino

Retrieving analog map...
Auto-discovery complete. Found 30 Digital Pins and 12 Analog Pins

Analog Call Input Callback: pin=2, Value=483 Time=2021-04-10 12:29:03 (Raw Time=1618072143.8835661)
Pin: 6 DHT Type: 22 Humidity:39.2, Temperature: 21.6 Timestamp: 2021-04-10 12:29:05
poll pin 6: humidity=39.2 temp=21.6 time of last report: 2021-04-10 12:29:07
```

Bingsitfx, 2021-04-10:
@MrYsLab Thank you so much it works!!

closed 2021-04-10 by Bingsitfx