MrYsLab/pymata4 #31: about the support for custom commands

mj2068, 2021-04-12:
after a few days of fiddling and testing around the pymata4 and FirmataExpress combo, I found this is indeed a powerful implementation of the Firmata protocol, and most importantly, it is actively maintained.
however, after having skimmed the user's guide and searched around, I didn't find a straightforward way to use custom commands.
can anybody give some guide regarding custom commands?
I didn't have a particular goal in mind, I just want to use python to program an Arduino, and want to have as total control as possible.
so, can you please confirm that I can't do that using the vanilla pymata4, right?
eg. if I had a customized FirmataExpress sketch uploaded to an Arduino, which has the ability to handle certain custom commands, I have to modify pymata4 in order to use it, right?

MrYsLab, 2021-04-12:
Thanks for the question. If you are adding a new feature, then both the Python side as well as the Arduino side need to be modified.
[This article](https://www.instructables.com/Going-Beyond-StandardFirmata-Revisited/) may provide some guidance to add a new feature.

Working with Firmata has a few challenges. First, because the serial link is in use, debugging can be very difficult. Also, the fact that FIrmata uses a 7-bit byte can add to the confusion of understanding how things work. Finally, for boards such as the Uno and Leonardo, there is very little space to add new features.

An alternative is [Telemetrix](https://mryslab.github.io/telemetrix/). It is very similar to Firmata, but does not use the 7-bit byte scheme, and was designed to be more easily extended. It has a couple of rudimentary debugging utilities that allow you to send variable or calculated values from the Arduino (or other supported MCUs) back to the Python client to be displayed in the console. Also, it is not as tight in available space. You can [read about it](https://mryslab.github.io/telemetrix/) here. There is a section in the user guide on [extending its 
functionality](https://mryslab.github.io/telemetrix/dht/). 

Currently, I have support for the Arduino AVR processors, STM32 (black pill), ESP8266 supporting WiFi, and the Raspberry Pi Pico. I also have a version that I am working on for the ESP32. WiFi for this version is working but I am trying to get Bluetooth to work as well. 

If you have any questions on either pymata4 or Telemetrix, please let me know.

MrYsLab, 2021-04-12:
I am going to close this issue, but if you have any further comments or questions you may post them here and I will see them.

closed 2021-04-12 by MrYsLab