MrYsLab/pymata4 #32: Error Exception in thread Thread-2:

ekokokokoko, 2021-05-19:
Hello, i have error exception in thread Thread-2:
how to fix it?
Note : I have firmata express & pymata4 updated to the latest version.
[image]

MrYsLab, 2021-05-19:
Thanks for the screenshot. To better understand what may be going on, could you please answer the following questions:

1. From the pin numbers it appears you are using a Mega2560. Is that a correct assumption?
2. What device types are you using - DHT11 or DHT22?
3. Does this happen immediately after starting the program? 
4. If not, does it happen after the same amount of time or does it just occur randomly?
5. Do you have any other sensors or actuators connected to the board?
6. I see that the sensor on pin 30 is returning 0.0 values. Is that what you expect? 
7. Is the sensor on pin 30 connected?
8. If you comment out the line in your code that establishes pin 30 as a DHT pin, do you still experience the problem.

Thanks.

ekokokokoko, 2021-05-19:
Thankyou for the answer.

1. Yes im using Mega2560 
2. DHT22
3. No
4. This error happen occur randomly, usually after 4 loop its give errors. The error  is also random, sometimes pin 30, sometimes pin 31, sometimes all 4 DHT sensors.
5. Yes, i also connected LDR sensor & 8 Relay module
6. No
7. Yes its connected properly
8. Yes im still experience error but with another sensor

After IndexError
all the sensor (DHT & LDR) gives the same results over and over again. I dont know why this is happen.
Another error, sometimes after the program launched gives error Firmata Sketch Firmware Version Not Found

Thanks

MrYsLab, 2021-05-19:
Thanks for the info. I only have 2 sensors to test with and am not able to reproduce what you are seeing. Here are some suggestions you might try:

1. Run [this example code ](https://github.com/MrYsLab/pymata4/blob/master/examples/dht.py)(you will need to modify the pin numbers) with 2 sensors. If you are not seeing the exception, add DHT sensors one at a time to the script to see if additional sensors cause the error.

2. If you are able to run the total number of DHT sensors you need without issue, I would then suggest modifying your code first to run the DHT sensors, and then add the LDR sensor. If that works, then add in the relay sensor.

The Firmata Sketch Firmware Version Not Found error usually occurs when a program is restarted without waiting enough time for the Arduino to reset itself. Waiting for the LEDs on Arduino to stop flashing upon powering up or resetting should solve this issue, and perhaps changing the arduino_wait parameter to be a longer time - from 4 to 6 seconds. Also, be aware that Analog sensors may continue streaming after the Python application is closed. The best thing to do here is performing a power-up reset of the Arduino.

Another item to try is to adjust the sampling period by calling [set_sampling_interval](https://htmlpreview.github.io/?https://github.com/MrYsLab/pymata4/blob/master/html/pymata4/index.html#pymata4.pymata4.Pymata4.set_sampling_interval) and give it a value of 30 milliseconds.

One last thing you may wish to try is to use [Telemetrix](https://mryslab.github.io/telemetrix/) instead of Firmata. There is a [DHT example](https://github.com/MrYsLab/telemetrix/blob/master/examples/dht.py) to try, and there is an [online API.](https://htmlpreview.github.io/?https://github.com/MrYsLab/telemetrix/blob/master/html/telemetrix/index.html)

Please let me know if any of this helps, and if not, which of the scenarios above are failing.

Thanks.

ekokokokoko, 2021-05-20:
Thanks for the response
i already try it, but give the same result with 4 dht sensors.
This is my code : 
"""
 Copyright (c) 2020 Alan Yorinks All rights reserved.
 This program is free software; you can redistribute it and/or
 modify it under the terms of the GNU AFFERO GENERAL PUBLIC LICENSE
 Version 3 as published by the Free Software Foundation; either
 or (at your option) any later version.
 This library is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 General Public License for more details.
 You should have received a copy of the GNU AFFERO GENERAL PUBLIC LICENSE
 along with this library; if not, write to the Free Software
 Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
"""

import time
import sys
from pymata4 import pymata4

"""
Setup a pin for dht mode
One pin is set for a dht22 and another for dht11
Both polling and callback are being used in this example.
"""

#
POLL_TIME = 5  # number of seconds between polls

# Callback data indices
CB_PIN_MODE = 0
CB_PIN = 1
CB_VALUE = 2
CB_TIME = 3

def the_callback(data):
    """
    A callback function to report data changes.
    This will print the pin number, its reported value and
    the date and time when the change occurred
    :param data: [pin, current reported value, pin_mode, timestamp]
    """

    tlist = time.localtime(data[5])
    ftime = f'{tlist.tm_year}-{tlist.tm_mon:02}-{tlist.tm_mday:02} ' \
            f'{tlist.tm_hour:02}:{tlist.tm_min:0}:{tlist.tm_sec:02}'

    print(f'Pin: {data[1]} DHT Type: {data[2]} Humidity:{data[3]}, '
          f'Temperature: {data[4]} Timestamp: {ftime}')

def dht(my_board, callback=None):
    """
     This function establishes the pin as a
     digital input. Any changes on this pin will
     be reported through the call back function.
     :param my_board: a pymata4 instance
     :param callback: callback funtion
     """

    # set the pin mode - for pin 6 differential is set explicitly
    my_board.set_pin_mode_dht(30, sensor_type=22, differential=.01, callback=callback)
    my_board.set_pin_mode_dht(31, sensor_type=22, differential=.01, callback=callback)
    my_board.set_pin_mode_dht(32, sensor_type=22, differential=.01, callback=callback)
    my_board.set_pin_mode_dht(33, sensor_type=22, differential=.01, callback=callback)

    # a flag to change the differential value after the first 5 seconds
    changed = False
    while True:
        try:
            time.sleep(POLL_TIME)

            # poll the first DHT
            value = board.dht_read(30)
            tlist = time.localtime(value[2])
            ftime = f'{tlist.tm_year}-{tlist.tm_mon:02}-{tlist.tm_mday:02} ' \
                    f'{tlist.tm_hour:02}:{tlist.tm_min:0}:{tlist.tm_sec:02}'
            print(f'poll pin 30: humidity={value[0]} temp={value[1]} '
                  f'time of last report: {ftime}')

            # format the time string and then print the data
            value = board.dht_read(31)
            tlist = time.localtime(value[2])
            ftime = f'{tlist.tm_year}-{tlist.tm_mon:02}-{tlist.tm_mday:02} ' \
                    f'{tlist.tm_hour:02}:{tlist.tm_min:0}:{tlist.tm_sec:02}'
            print(f'poll pin 31: humidity={value[0]} temp={value[1]} '
                  f'time of last report: {ftime}')

            # poll the Third DHT and print the values
            value = board.dht_read(32)
            tlist = time.localtime(value[2])
            ftime = f'{tlist.tm_year}-{tlist.tm_mon:02}-{tlist.tm_mday:02} ' \
                    f'{tlist.tm_hour:02}:{tlist.tm_min:0}:{tlist.tm_sec:02}'
            print(f'poll pin 32: humidity={value[0]} temp={value[1]} '
                  f'time of last report: {ftime}')
                        # poll the Third DHT and print the values
            value4 = board.dht_read(33)
            tlist = time.localtime(value4[2])
            ftime = f'{tlist.tm_year}-{tlist.tm_mon:02}-{tlist.tm_mday:02} ' \
                    f'{tlist.tm_hour:02}:{tlist.tm_min:0}:{tlist.tm_sec:02}'
            print(f'poll pin 33: humidity={value[0]} temp={value[1]} '
                  f'time of last report: {ftime}')
            
            print("    ")
            if not changed:
                # explicitly change the differential values
                my_board.set_pin_mode_dht(30, sensor_type=22, differential=20.0, callback=callback)
                my_board.set_pin_mode_dht(31, sensor_type=22, differential=20.0, callback=callback)
                my_board.set_pin_mode_dht(32, sensor_type=22, differential=20.0, callback=callback)
                my_board.set_pin_mode_dht(33, sensor_type=22, differential=20.0, callback=callback)
                changed = True
        except KeyboardInterrupt:
            board.shutdown()
            sys.exit(0)

board = pymata4.Pymata4()

try:
    dht(board, the_callback)
except KeyboardInterrupt:
    board.shutdown()
    sys.exit(0)

MrYsLab, 2021-05-20:
Thanks for supplying the code. I do not see anything wrong with the script.
I went through my code again to make sure that the DHT data is protected with a thread lock whenever the data is read or written, and it is.
I am still at a loss to understand why you are seeing this behavior.

I don't know if you have already done this or not, but if you run the script, starting with a single sensor enabled, and then run again, enabling two sensors, and continue to add sensors until you see the error, how many sensors can you run without error? If the error happens with 3 or 4 sensors but not 2, I will order some sensors to try and reproduce things here.

One other thing to try is to not poll the values but only have the callback print the values out. Does that work without error?

Thanks for your patience.

EkoWibowo1999, 2021-05-20:
@MrYsLab 
Sorry I am using a different account.
I've done what you suggested.  3 sensors can work properly, without issuing error. but when 4 sensors are connected, the program generates an error (Exception in thread Thread-2) , this usually happen after 4 to 6 loop. i didn't connect any other sensors to the mega except DHT22.

I also tried to changing the arduino_wait parameter to 6 seconds, But sometimes (not often) gives an error Firmata Sketch Firmware Version Not Found. As you said, this is resolved by performing a power-up reset of the Arduino.

Thank you, I hope this problem can be resolved

MrYsLab, 2021-05-20:
Thanks, that is very helpful. I will order some more sensors so I can try to reproduce the problem here. This may take several days to arrive. 

Did you have a chance to remove the polling statements and only use the callback? If so, does that also generate an error when using 4 sensors?

MrYsLab, 2021-05-20:
@ekokokokoko @EkoWibowo1999  Just sending a message to both accounts. Forget to add them to the previous message.

MrYsLab, 2021-05-20:
@EkoWibowo1999 @ekokokokoko Another thought. 
Could you try putting a small delay after each poll. 0.01 seconds should be adequate. Thanks.

EkoWibowo1999, 2021-05-20:
Okey, tomorrow i will remove the polling statements and only use the callback tomorrow and try putting 0.01 seconds delay after each poll. if it's still an error, I'll let you know.
Thanks

MrYsLab, 2021-05-22:
I found some DHT11's so I ran two DHT11's and two DHT 22's and I have now seen the error. On my system, it happens very rarely, but I will try to track down the cause. It appears that the FirmataExpress sketch is returning bad data, and since this happens rarely and randomly, my guess is that there is some timing/race condition occurring. I will post here when I figure out what is going on and a fix is ready.
Thanks for your patience.

EkoWibowo1999, 2021-05-22:
Glad to hear that. I run this program on raspberry pi 2B+ and this error happens every time it is launched. After that, sensor data is always the same
Example:
Poll pin 32 Temp 23
Error exception in thread Thread-2
(Error message)
Poll pin 32 Temp 23
Poll pin 32 Temp 23
Poll pin 32 Temp 23
And so on
Maybe if you use 4 dht22, it will happen same as me.

I hope this problem can be fixed
Thankyou

MrYsLab, 2021-05-22:
Thanks for the info. I see what the problem is but don't understand yet how it can happen. When data is received from the Arduino, it is placed on a Python deque. For SysEx commands, all the data should be captured and pulled of the deque before calling the method that processes the SysEx report data. However, occasionally, when a DHT report packet is presented for processing, it is shorter than it should be. When I add code to trap a short packet and set a breakpoint, I can see the short packet, but when I check the deque where data is placed for processing, the rest of the packet is in the deque. It should not be there and should have been removed from the deque before the DHT report processing called. There is some race condition that I do not yet understand. The problem is on the Python side and not the Arduino side as I originally thought.
Hopefully, I can figure out what is going on.

MrYsLab, 2021-05-22:
I figured it out, and it is nasty. The data being sent back from the DHT occasionally contains 247, which is also the value for the END_OF_SYSEX value. This was cutting things short. I am going to have to convert all the data from the DHT to 7-bit bytes. 

This will take me a day or two to do and to test. Hopefully, that will solve the issue and the mystery.

EkoWibowo1999, 2021-05-23:
Thanks for the update. I would appreciate it if you could solve this problem soon.

MrYsLab, 2021-05-23:
I am typing as fast as I can ;-). 
My code is based on the [DHTNew library](https://github.com/RobTillaart/DHTNew). Although this is not the cause of the issue, the code has changed quite a bit since I last ported it. What I want to do is to not port the library but actually to include and use it within FirmataExpress. If improvements are made to the library in the future, the Arduino IDE will provide notification, and the user can then update it without me having to change my code. I don't think this is a major job, but I will need to think it through, then code and test it. Both pymata4 and FirmataExpress need to be modified.
Assuming I don't run into any crazy problems, I should have this done by the end of the week but probably sooner.
Thanks again for your patience.

EkoWibowo1999, 2021-05-23:
I'm so sorry i didn't mean to rush you,i said so because your pymata4 is used to control the plant factory for my undergraduate thesis and scheduled to run on May 27th (Thursday). 

In the meantime, if i use telemetrix will it be the same error as pymata 4 ? If yes, what you suggestion to use for my plant factory? 

Thankyou

MrYsLab, 2021-05-23:
@EkoWibowo1999 @ekokokokoko 
No problem. I have a fix for you to try. I decided not to incorporate the library itself since the latest version cannot read DHT11 devices. I modified pymata4.py and the FirmataExpress. 

I will do a continuous run here for a few hours to make sure there are no more exceptions.

The pymata4 changes are in the [dht_fix branch](https://github.com/MrYsLab/pymata4/tree/dht_fix). You can use setup.py, or if you are comfortable with a pip3 local install, that will work as well.

The FirmataExpress changes are also in their [dht_fix branch.](https://github.com/MrYsLab/FirmataExpress/tree/dht_fix) If you bring up the FirmataExpress example in the Arduino IDE, overwrite it with [the new file](https://github.com/MrYsLab/FirmataExpress/blob/dht_fix/examples/FirmataExpress/FirmataExpress.ino), compile/upload, and you should be good to go.

Hopefully, this will solve the issue. Please let me know the results, good or bad.
Thanks.

EkoWibowo1999, 2021-05-24:
Thank you so much for your help.
i already try your new pymata4 & firmata express. about 30 minutes and no exceptions occured. But the temperature is wrong.

[image]
Above is running on arduino mega itself.

[image]
and here its run on arduino and raspberry pi 2B with your pymata4.

[image]
Thermohygrometer

My code is the same from last time.
My code : https://pastebin.com/MiDxTyB2

MrYsLab, 2021-05-24:
When I ran my tests the values printed correctly. I will test with your code and get back yo you later today.

MrYsLab, 2021-05-24:
Ok. Found it. Download again from the dht_fix repository. I suggest using pip3 to install this time. For some reason, when I used setup.py it did not overwrite with the new changes. There was a definite bug, but the values displayed in both Ubuntu and Windows looked correct. The "strange" values only appeared when I ran on Raspberry Pi.

If you are unfamiliar with pip3, here is what you need to do:
1. After downloading and extracting the repository go to the directory above where you placed pymata4-dht_fix. So for example, if you place it in /home/pi, then go to /home/pi.
2. Type:
```
sudo pip3 install ./pymata4-dht_fix
```
If you need to make changes and reinstall the use the following line:
```
sudo pip3 install ./pymata4-dht_fix --upgrade
```
Make sure that you have ./ in front of pymata4-dht_fix to choose the local directory.

I will be out this afternoon for several hours. I will check my mail upon returning in case you have any issues.

Let me know if things are now working or not.

Thanks

EkoWibowo1999, 2021-05-24:
I uninstall pymata4 first using pip uninstall pymata4. Then reinstall it using pip install. Now pymata4 version is 1.11.
When i run the program for couple hours, there are no exception but the temperature data is wrong. I must divided by 10 to make correct result.

Thankyou

MrYsLab, 2021-05-24:
If you can do the division in your code that would be great. The dhtNew library has become very unstable so I will moving over to dhtStable which is written by the same author of dhtNew. I won't be able to complete this before you have to submit your project, but I will be updating both pymata and telemetrix if dhtStable proves to work well.

EkoWibowo1999, 2021-05-24:
I will try the division tomorrow because it's 9pm now and the lab is closed. 

I'll let you know if my code is working or not
Thanks

MrYsLab, 2021-05-24:
I tried the code on the Raspberry Pi again, and I see the correct temperatures. If you would like to make changes within the pymata4.py code to adjust the temperature, you would either change the multiplier or add code to adjust[ at this line](https://github.com/MrYsLab/pymata4/blob/94eff3ecf8249415d597d1e12ad3b196990c17b8/pymata4/pymata4.py#L1511).

EkoWibowo1999, 2021-05-25:
I run this code sudo pip3 install ./pymata4-dht_fix and sudo pip3 install ./pymata4-dht_fix --upgrade, but still give the same result. What im missing? 
[image]
i decided to divide the temperature by 10 to make correct result. I will run my code for one day to make sure there are no more exception.

MrYsLab, 2021-05-25:
I did not make any more changes but provided a link to the source code if you wanted to do the adjustment in pymata4 instead of your own code. The choice is yours and probably easier done in your code.

EkoWibowo1999, 2021-05-27:
I've been running my code for 2 days, no exception error occurred. Thankyou so much.

But i have trouble with error firmata. I have changed Arduino Wait to 5 and 6 seconds but sometimes the error still occurs.

MrYsLab, 2021-05-27:
Most likely, the error is occurring because the Arduino is continuing to send data even though you have stopped and restarted the Python code. The Arduino does not provide for a hardware reset from a software command, so if you restart the Python code, you should also reset the Arduino, wait for the board LEDs to settle down, and then restart the Python code.

Just to let you know, I will be replacing the current DHT code to support the DHTstable library. The current version has bugs, but the author will be releasing an updated version soon. When he does, I will be updating pymata4, pymata-express, and FirmataExpress to use this library. It is much more stable than the current code.

EkoWibowo1999, 2021-05-28:
I made a new python code to check whether the main code is running or not, if not (Error Firmata) then restart the raspberry.

There is another problem. sometimes the sensor readings show "-1". I check the sensor wiring on arduino nano, its working properly, so its not wiring problem. I don't know why. To overcome this, I programmed (on main code) to restart if the result had been repeated 5 times.

Thankyou for your information, i hope the new DHTstable library has no bugs and more stable.

MrYsLab, 2021-05-28:
digitalWrite 0 sets the pin to ground, and 1 sets it to either 5v or 3v depending upon the board.

I am not going to implement the robotdyn dimmer. However, if there is an Arduino Core library that supports it, I would suggest adding it to [Telemetrix4Arduino](https://github.com/MrYsLab/Telemetrix4Arduino).
Telemetrix was designed so that users can extend its functionality in a much simpler manner than that required to extend firmata. I would be happy to provide any guidance if you decide to do this.

Also, I will be updating telemetrix for the DHT since that too is somewhat broken. I am having some issues with it, but hope to have it completed within about a week. 

I will be publishing the changes for pyamat4 as soon as the latest DHTstable library becomes available in the Arduino library manager. Currently they still have the old version.

MrYsLab, 2021-05-28:
As soon as I publish the changes using the new library. I will be closing this issue, however, you can still post here after it is closed and I will be notified.

MrYsLab, 2021-05-30:
This is resolved with release 1.11 and requires FirmataExpress version 1.2.0.

closed 2021-05-30 by MrYsLab