MrYsLab/pymata4 #36: Registry for i2cwrite missing

theputernerd, 2021-07-31:
i2c_write doesn't seem to work as there is no way to provide the registry of the device. (I am new to I2C protocol), but https://www.programmersought.com/article/36315758270/ shows that the registry is supposed to be a parameter for writing to a device.

MrYsLab, 2021-07-31:
@theputernerd 
Hi. I think you may have meant register and not registry. If this is not correct, could you please explain what you mean by registry since I am not familiar with that term in relation to i2c?

If, in fact, you meant register, the registers are defined by the datasheet for the device, and each i2c device has its own set of registers.

As an example, if you look at the ADXL345  code I provided, you will see that the power control register is being programmed [here.](https://github.com/MrYsLab/pymata4/blob/15f4eea7627e047cfb6a31574bd43f62558f30d5/examples/i2c_adxl345_accelerometer.py#L44) The code is performing an i2c write with the address of the device being 83 and two bytes being written to the device, that is 45 and 0. The value of 45, which is 0x2D in hex, is the power control register value, and 0 is what that register is being set to. If you look at the [spec sheet for the ADXL345](https://www.analog.com/media/en/technical-documentation/data-sheets/ADXL345.pdf), on page 23, there is a list of all the registers. If you scroll down further in the document, it explains how to set the various bits within the register.

So the i2c support that pymata4 provides is not specific to an i2c device, but is a general programming interface allowing control and manipulation of the device's registers. 

i2c programming is not always simple, and if you are using a device that has an Arduino library that supports it, understanding how that library is manipulating the registers is a great way to learn what needs to be done in porting that library to the pymata4 API.

If any of this is unclear, or if you have any other questions, please let me know.
Thanks for the question.

MrYsLab, 2021-07-31:
I am closing this issue, but if you have any other questions or if you think it should remain open, please post a message here.
Thanks.

closed 2021-07-31 by MrYsLab