MrYsLab/pymata4 #42: Polling with sonar HC-SR04

dlefur, 2021-11-10:
Hello.

First, a big thank for your work. It is extreamely easy to use. A huge upgrade for prograaming with Arduino !

When polling without any callback, with the HC-SR04, the measured distance is always 0.

Cordially.

PS : have you worked with the ccs811 sensor ?

**The programm that returns 0 :**
```python
from pymata4 import pymata4
import sys
import time
carte = pymata4.Pymata4()
print("--------------- Début --------------")
# Constantes
Delai=1
Trigger_pin = 8 # port du Trigger D8
Echo_pin = 7    # port du Echo D7
#----------------  programme principal
try :
    carte.set_pin_mode_sonar(Trigger_pin,Echo_pin)
    time.sleep(1)
    while True:
        time.sleep(Delai)
        (dist,temp)=carte.sonar_read(Trigger_pin)
        print(f"Distance : {dist} cm")
except KeyboardInterrupt :
    print("--------------- Fin --------------")
    carte.shutdown()
    sys.exit(0)
```
with a callback, it works : 

**The program that works :**
```python
from pymata4 import pymata4
import sys
import time
carte = pymata4.Pymata4()
print("--------------- Début --------------")
# Constantes
Delai=1
Trigger_pin = 8 # port du Trigger D8
Echo_pin = 7    # port du Echo D7

#------------------ fonction callback
def nouvelle_distance(data):
    dist=data[2]
    #print(f"Nouvelle distance : {dist} cm")

#----------------  programme principal
try :
    carte.set_pin_mode_sonar(Trigger_pin,Echo_pin,callback=nouvelle_distance)
    time.sleep(1)
    while True:
        time.sleep(Delai)
        (dist,temp)=carte.sonar_read(Trigger_pin)
        print(f"Distance : {dist} cm")
except KeyboardInterrupt :
    print("--------------- Fin --------------")
    carte.shutdown()
    sys.exit(0)
```

MrYsLab, 2021-11-10:
I will need to try and recreate this and will get back to you. The ccs811 is an i2c device. You can use it, but you would need to write the i2c code. An example that demonstrates using an i2c accelerometer may be viewed [here](https://github.com/MrYsLab/pymata4/blob/master/examples/i2c_adxl345_accelerometer.py).

Because pymata4 has essentially been replaced with [Telemetrix](https://mryslab.github.io/telemetrix/) you may wish to check it out. It also requires Python 3.7, so you will have the same issue with Python 3 versions.

MrYsLab, 2021-11-10:
This issue is fixed with release 1.14. Thanks for bringing this issue to my attention. I am closing the issue, but if it is not resolved, please leave a comment here, and I can reopen it.

closed 2021-11-10 by MrYsLab

dlefur, 2021-11-10:
Thanks a lot for this reactivity ! I am going to explore telemetrix now.