MrYsLab/pymata4 #43: Bluetooth supports

Creepercdn, 2021-11-20:
In particular, I am using ZS-040.

MrYsLab, 2021-11-20:
Thank you for your suggestion. Unfortunately, the FirmataExpress sketch is low on space for the Arduino Uno, and therefore, I cannot add any new features.

closed 2021-11-20 by MrYsLab

Creepercdn, 2021-11-21:
It doesn't need to modify anything in FirmataExpress, just add Bluetooth supports in pymata4 side. Read serial data in Bluetooth with a library (like pybluez), and parse it like using USB serial port.

Creepercdn, 2021-11-21:
so, i tested for it for 10 hours, modified pymata4 to support pybluez library, and found it only working normal with StandardFirmata.

Creepercdn, 2021-11-21:
https://github.com/Creepercdn/pymata4

MrYsLab, 2021-11-21:
Thanks for your efforts, and I am glad you successfully got things to work for your environment.
However, I will not be incorporating it into this repository. So that you don't feel I've randomly denied the merge, here is a list of reasons why. I hope that you will understand.
1. For Windows, pybluez requires a whole [C++ development _environment](https://github.com/pybluez/pybluez/blob/master/docs/install.rst#windows-dependencies)  which is a significant burden for some users.
2. The changes need to be tested against every feature using [the examples](https://github.com/MrYsLab/pymata4/tree/master/examples) for both Linux and Windows.
3. For consistency between pymata4 and[ pymata-express](https://github.com/MrYsLab/pymata-express), an asyncio version needs to be written and tested.
4. The documentation needs to be updated to reflect the changes.
5. I will be archiving pymata4 and pymata-express in the very near future, meaning that the features are frozen, and the only development will be for bug fixes.

I will be working on BLE support for [Telemetrix](https://github.com/MrYsLab/telemetrix) and will consider support for Classic Bluetooth at that time as well. I cannot promise when or if I will get this done. I attempted coding Bluetooth last year and faced many challenges, which I documented [here](https://mryslab.github.io/bots-in-pieces/bluetooth/2021/01/28/bluetooth-blues.html). I do not know if I can locate Python Bluetooth libraries that work for Linux and Windows and don't require the user to have compilation tools sets installed on their computer.

Again, thanks for your efforts.

Creepercdn, 2021-11-22:
I hope your work can be successful.